Domochip/WPalaControl #1: Hardware ?

Aorlhac, 2020-12-05:
Hello, do you still have some PCB and box left ? i'm very interested 👍

Domochip, 2020-12-05:
Yes, you can contact me directly at domochip[@]hotmail.com

djpostka, 2021-05-17:
Hello,

I will also wish to buy a pcb if possible. thank you.

Domochip, 2021-05-17:
@djpostka I just found your email into junks, I'm replying directly through mail

nargotik, 2022-12-09:
HI @Domochip just sent you and email to check if you still have some kits

Domochip, 2022-12-12:
Hi, Yes, I have, let's continue by mail :-)

starship666, 2023-12-10:
Hi @Domochip, can't find your email but I'm also interested if you still have some kits to try it.

frisk2, 2024-02-13:
Hello, do you still have some PCB and box left ? i'm interested.

Domochip, 2024-02-13:
Hi, Yes, I have, let's continue by mail :-)
domochip[@]hotmail.com

Domochip, 2024-09-12:
closing this issue because tindie product is now available: https://www.tindie.com/products/domochip/wirelesspalacontrol-aka-wpalacontrol/

closed 2024-09-12 by Domochip