Domochip/WPalaControl #4: PCB Control Palazetti D1

Polux01, 2021-11-04:
Bonjour, 

Je me permet de vous écrire car je suis de très près votre projet qui permet de remplacer le connection box de palazetti, il y a 5 ans que j'ai mon poele et celui ci est compliqué pour madame... J'avais pour idée de faire comme vous mais mes maigres compétence en codage et electronique m'ont vite limités.
Votre projet me remotive donc à domotiser ce poele. 
Si il vous reste des PCB je veux bien vous en acheter un ou deux. 
Cordialement
Et merci pour ce formidable travail

Domochip, 2021-11-07:
Bonjour,

Réponse faite par mail directement ;-)

Domochip, 2021-11-08:
Bonjour,

Vous pouvez me contacter directement à l'adresse suivante domochip[@]hotmail.com

closed 2021-11-09 by Domochip