Domochip/WPalaControl #6: Wifi suggestion

Flobul, 2022-02-26:
Hi.

I have 3 mesh wi-fi routers.
When I select the SSID, I can't figure out which is the nearest router.
[image]
Show BSSID or signal strength ?

Domochip, 2023-12-29:
Hi,

The configuration contains the ssid of the network. In that case, no differences between two Livebox SSID.
The problem is that I cannot find any info about ESP8266 behavior if multiple network are available with the same SSID.
I expect it to connect to the strongest AP, but maybe I'm wrong.
Did you observed a different behavior?

Flobul, 2023-12-31:
I assigned WPalaControl to a specific mesh router, I haven't encountered this bug since.

Before: When I disconnected the router to which WPalaControl was assigned, WPalaControl would connect to another remote router (with low RSSI) and remain connected to it even after the main router had been restarted.
But I did not reproduced this bug so far. Once the main router came back, the WPalaControl is assigned to it.

Maybe related to a mesh router issue.

So feel free to close this one ;)

closed 2023-12-31 by Domochip