Domochip/WPalaControl #7: Wifi backup?

Flobul, 2022-02-26:
Sometimes, I reboot my wi-fi router and the Wpalacontrol does not reconnect to my router, it creates its own AP, and never try to reconnect. So I have to reset it manually. I notice only when I try to launch manually... So maybe a few days later.

Maybe adding a second SSID network (fallback)? I the first is unreachable, try the other one?
Or maybe a watchbog: if in AP because SSID network was unreachable, retry in 5 minutes?

Domochip, 2022-02-26:
Very strange, WirelessBase should look back for configured SSID every 60sec (during 20sec).
It does only if there is NO Client currently connected to the AP.
Is it your case?

Flobul, 2022-02-26:
No client connected to the WpalaControl AP.

carmelo42, 2022-08-30:
Same here, when the router is restarted, I can see the the Wpalacontrol AP, but even if no client is connected it does not reconnect to my router. I have to reboot Wpalacontrol.

Domochip, 2022-08-30:
please, ensure that no laptop or mobile phone connect to the module

carmelo42, 2022-08-31:
Yes, you were right ... I've forgotten a laptop used once to configure the module , my bad !
Sorry for the noise.

Flobul, 2022-11-11:
Nevermind, an old phone I used to configure WPalaControl was connecting to it...

closed 2022-11-11 by Flobul