Domochip/WPalaControl #13: Compatibility with ET4W control

mattiabrunello5, 2022-09-15:
Hi, is hass-palazzetti compatible with et4w control ( for fan speed and combustion?

Thanks for answer

mattiabrunello5, 2022-09-15:
hi i make a mistake, sorry, (haas-palazzetti si another repository )

ET4W is another controller of palazetti for control of fireplaces, its work on wifi, i link above a page for more info:
https://palazzettigroup.com/research-and-development/technology/

Domochip, 2022-09-15:
Does Easytech4wood is embedded into the stove or this is an option?

mattiabrunello5, 2022-09-15:
Is an option for pellet stove 
Wood stove 
And fireplace

Domochip, 2022-09-15:
Then it depends of the motherboard, if it matches with the one in WPalaControl description then it will work otherwise I'll need more info of the new motherboard.
Description: https://github.com/Domochip/WirelessPalaControl#compatibility

mattiabrunello5, 2022-09-15:
This is a link of the manual of this interface 

https://cdn.palazzetti.it/manuals/004779060.pdf

Domochip, 2022-09-15:
I was talking about the stove motherboard

mattiabrunello5, 2022-09-15:
I don't have the possibility to check  ,but maybe it's the same ?

mattiabrunello5, 2022-09-15:
I noticed a electric wiring Diagram il this pdf

https://cdn.palazzetti.it/manuals/004778071.pdf

https://cdn.palazzetti.it/manuals/004777731.pdf

Domochip, 2022-09-16:
Sorry, but no docs give enough about the motherboard.
I can't ensure that WPalaControl willl be able to connect to it.

mattiabrunello5, 2022-09-16:
ok thanks

closed 2022-09-16 by mattiabrunello5