Domochip/WPalaControl #15: Configuration and infos on TIMER status

1300371, 2022-10-06:
Hello,

I've got a Palazzetti stove and there is a TIMER fonction that you have to turn on if you want to schedule ON/oFF.

Is there a way to get that _TIMER status_ and also to set it ?

Here is a video of the feature: https://www.youtube.com/watch?v=YBgKCtgVSOY

Domochip, 2022-10-06:
Hi, I never used it, but I think this is `Chrono Data`/`Chrono Status` that you are looking for.
And `Chrono Program` is the way to configure its planning.

1300371, 2022-10-06:
BTW, is it possible to forward more data via mqtt ? All these data are availble only via http ? 😕

1300371, 2022-10-10:
Ping @Domochip

1300371, 2022-10-24:
Did read you my second question @Domochip ? Sorry for spam

Domochip, 2023-12-28:
Hello,
Sorry for the very very late reply.
The new V3.0.0 (just released) allows you to get this info via MQTT if you send "GET+CHRD" to cmd topic.
Could you give it a try and let me know if it matches to your needs?

1300371, 2023-12-29:
No problem.  
To update, just have to upload the _bin_ firmware file througth the UI ?

Domochip, 2023-12-29:
yes

Domochip, 2024-01-30:
Ping @1300371

1300371, 2024-01-30:
Sorry 😅
As this is winter in France, I want to wait spring to not jeopardize my installation (in case of something goes wrong)

closed 2024-02-18 by Domochip