Domochip/WPalaControl #16: Any chance update code for ESP32 ?

Flobul, 2023-01-15:
Hi.

Sorry to bother, but is there any chance you update this repo for another hardware plateform? (espressif32 instead of espressif8266)
Or maybe a way to use the lib with another system (tasmota, espeasy...)

Domochip, 2023-01-15:
Hi,

mm, regading other systems, it is difficult to respect right timing and listen on UDP + TCP + Web like official box.

For ESP32, I tried a few years ago to port it but Ticker class was missing. It seems to exists now and this option seems possible.
ESP32 Mini are still larger than D1Mini and cost a little more so the right question is Why? Do you see a particular improvement?

Domochip, 2024-09-04:
Hello,

code has been adapted to ESP32 on test-esp32 branch

Domochip, 2024-09-04:
And tested successfully on my side

closed 2024-09-07 by Domochip

Flobul, 2024-09-23:
Thank you! Nice job.

I made it work but got some compiling errors in ESP8266:
```
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Indexing .pio/build/d1_mini/lib64f/libPubSubClient.a
In file included from src/base/Application.cpp:1:0:
src/base/Application.cpp: In lambda function:
src/base/Application.h:9:41: error: 'using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' has no member named 'keepAlive'
 #define SERVER_KEEPALIVE_FALSE() server.keepAlive(false);
                                         ^
src/base/Application.cpp:165:5: note: in expansion of macro 'SERVER_KEEPALIVE_FALSE'
     SERVER_KEEPALIVE_FALSE()
     ^
src/base/Application.cpp: In lambda function:
src/base/Application.h:9:41: error: 'using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' has no member named 'keepAlive'
 #define SERVER_KEEPALIVE_FALSE() server.keepAlive(false);
                                         ^
src/base/Application.cpp:173:5: note: in expansion of macro 'SERVER_KEEPALIVE_FALSE'
     SERVER_KEEPALIVE_FALSE()
     ^
src/base/Application.cpp: In lambda function:
src/base/Application.h:9:41: error: 'using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' has no member named 'keepAlive'
 #define SERVER_KEEPALIVE_FALSE() server.keepAlive(false);
```

```
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_waveform_pwm.cpp.o
src/WirelessPalaControl.cpp: In member function 'bool WebPalaControl::executePalaCmd(const String&, String&, bool)':
src/WirelessPalaControl.cpp:933:28: error: 'class ESP8266WiFiClass' has no member named 'getHostname'
       data["LABEL"] = WiFi.getHostname();
                            ^
Compiling .pio/build/d1_mini/FrameworkArduino/core_esp8266_wiring.cpp.o
src/WirelessPalaControl.cpp:951:28: error: 'class ESP8266WiFiClass' has no member named 'broadcastIP'
       data["WBCST"] = WiFi.broadcastIP().toString();
```

```
src/WirelessPalaControl.cpp: In member function 'bool WebPalaControl::executePalaCmd(const String&, String&, bool)':
src/WirelessPalaControl.cpp:933:28: error: 'class ESP8266WiFiClass' has no member named 'getHostname'
       data["LABEL"] = WiFi.getHostname();
                            ^
src/WirelessPalaControl.cpp:951:28: error: 'class ESP8266WiFiClass' has no member named 'broadcastIP'
       data["WBCST"] = WiFi.broadcastIP().toString();
                            ^
```

Here are the changes I made to "fix" it:
https://github.com/Flobul/WirelessPalaControl/pull/3/files

Domochip, 2024-09-23:
everything has been merged on WPalaControl master branch and I'm able to compile ESP32 version by just switching to the esp32 env: 
[image]

Do you have a different result?

Flobul, 2024-09-23:
esp8266 (KO):
```
 *  Exécution de la tâche : platformio run 

Processing d1_mini (framework: arduino; board: d1_mini; platform: espressif8266)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing domochip/Palazzetti#v2.7.3
Unpacking  [####################################]  100%
Library Manager: Palazzetti@2.7.3 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
---- Copying base files ----
----------------------------
--- Converting web files ---
----------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2023.4.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.4+9 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing domochip/Palazzetti#v2.7.3
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80%
Library Manager: Palazzetti@2.7.3 has been installed!
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.2.0
|-- PubSubClient @ 2.8.0
|-- EEPROM @ 1.0
|-- ESP8266WebServer @ 1.0
|-- ESP8266WiFi @ 1.0
|-- LittleFS(esp8266) @ 0.1.0
|-- Palazzetti @ 2.7.3
|-- Ticker @ 1.0
|-- ESP8266mDNS @ 1.2
Building in release mode
Compiling .pio/build/d1_mini/src/WirelessPalaControl.cpp.o
Compiling .pio/build/d1_mini/src/base/Application.cpp.o
Compiling .pio/build/d1_mini/src/base/Core.cpp.o
Compiling .pio/build/d1_mini/src/base/MQTTMan.cpp.o
Compiling .pio/build/d1_mini/src/base/Main.cpp.o
Compiling .pio/build/d1_mini/src/base/Utils.cpp.o
Compiling .pio/build/d1_mini/src/base/WifiMan.cpp.o
Generating LD script .pio/build/d1_mini/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/d1_mini/lib075/PubSubClient/PubSubClient.cpp.o
Compiling .pio/build/d1_mini/lib248/EEPROM/EEPROM.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiGratuitous.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
.pio/libdeps/d1_mini/PubSubClient/src/PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio/libdeps/d1_mini/PubSubClient/src/PubSubClient.cpp:523:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     return (rc == expectedLength);
                   ^
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Archiving .pio/build/d1_mini/lib248/libEEPROM.a
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pio/build/d1_mini/lib359/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
In file included from src/base/Application.cpp:1:0:
src/base/Application.cpp: In lambda function:
src/base/Application.h:9:41: error: 'using ESP8266WebServer = class esp8266webserver::ESP8266WebServerTemplate<WiFiServer>' has no member named 'keepAlive'
 #define SERVER_KEEPALIVE_FALSE() server.keepAlive(false);
[... 401 lines cut ...]
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/mhetesp32minikit/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/mhetesp32minikit/libFrameworkArduino.a
Indexing .pio/build/mhetesp32minikit/libFrameworkArduino.a
Linking .pio/build/mhetesp32minikit/WirelessPalaControl.32.elf
Retrieving maximum program size .pio/build/mhetesp32minikit/WirelessPalaControl.32.elf
Checking size .pio/build/mhetesp32minikit/WirelessPalaControl.32.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.7% (used 41488 bytes from 327680 bytes)
Flash: [======    ]  55.4% (used 725809 bytes from 1310720 bytes)
Building .pio/build/mhetesp32minikit/WirelessPalaControl.32.bin
esptool.py v4.7-dev
Creating esp32 image...
Merged 26 ELF sections
Successfully created esp32 image.
====================================================== [SUCCESS] Took 15.54 seconds ======================================================

Environment       Status    Duration
----------------  --------  ------------
mhetesp32minikit  SUCCESS   00:00:15.543
======================================================= 1 succeeded in 00:00:15.543 =======================================================
 *  Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer. 

```

Domochip, 2024-09-23:
It seems I'm using a newer platform and framework for esp8266 than yours: 
[image]

Could you give it a try first.

Flobul, 2024-09-23:
Done and fixed. framework-arduinoespressif8266 is in platform Espressif 8266.
I had many version of Espressif 8266 installed. I removed oldest to keep 4.1.0 and 4.2.0 versions, and got it fixed.
Thanks.