Domochip/WPalaControl #18: Stove infos

Ir0n42, 2023-04-06:
Hi, I have received my controller with 1.4.1 system running.
I have configured it on my network and plug RJ11 on my Palazzetti Monica. 
But now, stove info still empty, why? Forgot something? 
Thank you

Domochip, 2023-04-06:
Hi, 
Could you provide a screenshot of the complete version number.
maybe a picture of the cabling to the stove too.

Thanks

Ir0n42, 2023-04-06:
Yes sure! 
[image]

Ir0n42, 2023-04-06:
[image]
[image]
[image]

Ir0n42, 2023-04-06:
[image]

Ir0n42, 2023-04-06:
[image]
[image]

Ir0n42, 2023-04-06:
[image]

Ir0n42, 2023-04-06:
[image]

Ir0n42, 2023-04-06:
[image]

Ir0n42, 2023-04-06:
[image]

Domochip, 2023-04-07:
OK, 
First I advise to update to the last firmware version.
Next to this one, you should see serial number on the landing page.
With your current version, even if nothing appears, it doesn't mean the module doesn't work.
You can try to query this URL to check : http://**{IP}**/cgi-bin/sendmsg.lua?cmd=GET+STDT

Ir0n42, 2023-04-07:
Ok thank you, I have update it. 
Message :
 Stove infos (07/04/2023 10:02:30)
    MSG : Stove communication failed! please check cabling to your stove.
And when I Go to http://{IP}/cgi-bin/sendmsg.lua?cmd=GET+STDT
Message : 
{"INFO":{"CMD":"GET STDT","MSG":"Stove communication failed","RSP":"TIMEOUT"},"SUCCESS":false,"DATA":{"NODATA":true}}

Domochip, 2023-04-07:
It seems you have an issue with the cabling.
Does the cable match with the image here ? : https://github.com/Domochip/WirelessPalaControl#connect

Ir0n42, 2023-04-07:
Yes ! It's crossed

Domochip, 2023-04-07:
Then, do you have a picture of your stove motherboard?

Ir0n42, 2023-04-07:
I will open it

Ir0n42, 2023-04-07:
That enouth for you? 
[image]

Ir0n42, 2023-04-07:
@
[image]

Ir0n42, 2023-04-07:
[image]

Ir0n42, 2023-04-07:
[image]

Domochip, 2023-04-07:
Yes, defenitly!
This is a MicroNova motherboard and WirelessPalaControl is **NOT** compatible with MicroNova.
As per written in first chapters (https://github.com/Domochip/WirelessPalaControl), this project aims to control Fumis based stoves.
Your stove should be pretty old to have this kind of Motherboard.

I let you contact me by mail for further discussion.

closed 2023-04-07 by Domochip