Domochip/WPalaControl #29: Set silent mode returns a timeout error

Flobul, 2024-01-21:
Hi.

When SET+SLNT+1 command is sent, the room fan stops, but the cmd returns a timeout issue.
When SET+SLNT+0 command is sent, nothing happens (is it supposed to recover last fan value ?)

```
Mac@Flobul % curl "http://192.168.23.49/cgi-bin/sendmsg.lua?cmd=SET+SLNT+0"
{"INFO":{"CMD":"SET SLNT","RSP":"OK"},"DATA":{"SLNT":0,"PWR":1,"F2L":0,"F2LF":0},"SUCCESS":true}%    

Mac@Flobul % curl "http://192.168.23.49/cgi-bin/sendmsg.lua?cmd=SET+SLNT+1"
{"INFO":{"CMD":"SET SLNT 1","RSP":"TIMEOUT","MSG":"Stove communication failed"},"DATA":{"NODATA":true},"SUCCESS":false}%   
```

Domochip, 2024-01-30:
Hi,

SET+SLNT+1 should not return a timeout error, this is effectively a bug.
Regarding SET+SLNT+0, it does nothing...

I'll have a look at this bug shortly.
thx

Domochip, 2024-02-04:
Hi, Could you give it a try with the latest pre-release?
If the error still occur, I'll need your GET+STDT to understand, what happens.

Flobul, 2024-02-04:
Hi.

```
Mac@Flobul % curl "http://192.168.23.49/cgi-bin/sendmsg.lua?cmd=SET+SLNT+0"
{"INFO":{"CMD":"SET SLNT","RSP":"OK"},"DATA":{"SLNT":0,"PWR":1,"F2L":0,"F2LF":0},"SUCCESS":true}%
Mac@Flobul % curl "http://192.168.23.49/cgi-bin/sendmsg.lua?cmd=SET+SLNT+1"
{"INFO":{"CMD":"SET SLNT 1","RSP":"TIMEOUT","MSG":"Stove communication failed"},"DATA":{"NODATA":true},"SUCCESS":false}%  
```

Initial values : FAN OFF and POWER 1. When I send SLNT+1, POWER is set to 3. Nothing happens on SLNT+0.

Domochip, 2024-02-04:
Hi,

First, thank you, following the track, I found a bug I introduced recently (I just fixed the pre-release).
Regarding your issue, The "SILENT" mode seems to be designed for Stove which have FAN3 and FAN4 (additionnal Air ducts or Water pumps).

Here is what it is doing: Set RoomFan, FAN3 and FAN4 to 0
[image]
Your Stove has FAN2TYPE = 2 which means you have only Room Fan and then iSetRoomFan3Atech return an error.

So for your model, you just need to do a SET+RFAN+0
I will improve error messages of WPalaControl in next release, then it will return an UNSUPPORTED error.

Flobul, 2024-02-04:
You're welcome.

Only ROOM FAN on my stove, exactly.

I just tested your latest fix, same cmd results than before, but now POWER is no more set to 3 anymore.

closed 2024-02-04 by Domochip