Domochip/WPalaControl #33: Add F3L and F4L setting

carmelo42, 2024-02-09:
Just tried the 2.1.5/3.1.2v2 pre-release : 

[image]

It works great :) 

I can change the main fan setting (F3L) but there are 2 settings missing : left and right fans (F3L and F4L) : 

[image]

Domochip, 2024-02-09:
Thank you for the feedback.
I had a doubt about this part because my stove doesn't have this additionnal fans.
So, if I understand well, it would be better to have 2 sliders for F3 and F4, right? With value from 0 to 5?

carmelo42, 2024-02-09:
Yes exactly :) 
2 sliders for the 2 additional fans, scale 0 to 5.

pm2036, 2024-02-09:
Hello, 
On my stove, F3L and F4L is only 0/1. Maybe it depends of the stove model?

[image]
[image]

carmelo42, 2024-02-09:
Mine is "TURBOFONTE TAL" which is I think the same as PALAZZETTI JULIA : 

```
// 20240209100452
// http://192.168.0.239/cgi-bin/sendmsg.lua?cmd=SET+FN4L+4

{
  "INFO": {
    "CMD": "SET FN4L",
    "RSP": "OK"
  },
  "DATA": {
    "F4L": 4
  },
  "SUCCESS": true
}
```
On the MQTT bus : 
[image]

And for the other fan : 
```
// 20240209100713
// http://192.168.0.239/cgi-bin/sendmsg.lua?cmd=SET+FN3L+4

{
  "INFO": {
    "CMD": "SET FN3L",
    "RSP": "OK"
  },
  "DATA": {
    "F3L": 4
  },
  "SUCCESS": true
}
```
[image]

pm2036, 2024-02-09:
Ok, mine is a Palazzetti Allegro

`http://192.168.110.70/cgi-bin/sendmsg.lua?cmd=SET+FN3L+4`
`{"INFO":{"RSP":"OK","CMD":"SET FN3L","TS":1707471245},"SUCCESS":true,"DATA":{"F3L":1}}`

I think the max value is in sendmsg.lua?cmd=GET+ALLS
"FANLMINMAX":[0,5,0,1,0,1]

Domochip, 2024-02-09:
Yes, FANLMINMAX contains these limits.
2 firsts numbers are min and max for RoomFan (aka FAN2)
2 nexts are for F3
then 2 last numbers are for F4

So, the good option id to use some switches for stove which have "0,1" limits and sliders for "0,5" limits
I'll implement that shortly

Domochip, 2024-02-09:
V3 prerelease is available and adjust the HomeAssistant Entity type based on the Fan Min and Max values.
Could you let me know if it fit your needs?

carmelo42, 2024-02-09:
It works great :) 

[image]

Domochip, 2024-02-09:
In your case, switch buttons can be removed then

carmelo42, 2024-02-09:
Yes I will remove them. 
I'm happy, I've removed https://github.com/marcopal74/home-assistant-palazzetti which was not working every time.

pm2036, 2024-02-09:
Ok for me too, thanks.
[image]

closed 2024-02-18 by Domochip