Domochip/WPalaControl #34: SET+PWR+2 sent instead of SET+POWR+2

carmelo42, 2024-02-11:
Hello,

From HA, when is use the slider to set fire power to 2, the command sent is SET+PWR+2 instead of SET+POWR+2 so it doesn't change the value:

[image]

If I send the correct command through HTTP it's ok: 
```
// 20240211090505
// http://192.168.0.239/cgi-bin/sendmsg.lua?cmd=SET+POWR+2

{
  "INFO": {
    "CMD": "SET POWR",
    "RSP": "OK"
  },
  "DATA": {
    "PWR": 2,
    "F2L": 2,
    "FANLMINMAX": [
      0,
      5,
      0,
      5,
      0,
      5
    ]
  },
  "SUCCESS": true
}
```

Domochip, 2024-02-11:
thx, I republished the v3 prerelease with to fixed topic

carmelo42, 2024-02-11:
Flashed, and it works fine :)
Thanks a lot :)

closed 2024-02-11 by carmelo42