Domochip/WPalaControl #35: Not working on ESP01

Christoffer-da, 2024-02-25:
I tried to flash and run on another esp8266 variant, the esp01.
Unfortunately it flashes well, but won't run. 
I get the rst error 4, boot mode 3.7.

Would be nice to be able to use this firmware on the different esp8266 variants. This could also involve the possibility to set rx/tx gpio in the ui.

Christoffer-da, 2024-02-25:
When I get the time, I will try to change the board in platform io, and also the pin-out, to see if I can get everything working.

Domochip, 2024-02-25:
This project has been designed on D1 Mini which is much more convenient than ESP01. What is the benefit to run it on ESP01?

Christoffer-da, 2024-02-25:
For a project that basically only uses rx-tx, the small and cheap esp-01 might be better suited. You do not loose function, and it is way cheaper.
I mostly use the esp32, but for this it would be great to reuse a modbus module I have previously used to control my underfloor heating system.

For me the d1mini is great, and if I do not get it up and running, I might just buy from your tindie shop, for the ease of it.

closed 2024-03-02 by Christoffer-da