Domochip/WPalaControl #36: CMD+ON while stove is burning: expected behavior?

sven337, 2024-03-02:
Hello,
this is probably not a bug, but I've noticed that if I start the stove with CMD+ON, then send another CMD+ON while it's in status "Fueling", it moves to "TestFire" status and stops feeding pellets, then stops burning.
Is that expected? Have you noticed that on yours?

Domochip, 2024-09-04:
Hello,

I really don't know if this is expected.
I didn't noticed it on mine because it is always ON so I never throw CMD+ON command :-D

sven337, 2024-09-08:
Mine is always on ON also, but it regulates at a low temperature. Then I use CMD+ON to force a burn when needed.

Domochip, 2024-09-22:
ok, there is nothing reflecting this in my Palazzetti Lib.
It seems to be a Fumis motherboard behavior/defect/feature ...