Domochip/WPalaControl #37: Add another rj11 connector on the board

Christoffer-da, 2024-03-12:
Just received the board, and now awaiting the last pieces, to be able to install. 

I do not know if this is a valid suggestion, but here we go:

If you add another RJ11 connector on the next revision of the board, bridged with the existing one, the we could save one cable and the external rj11 splitter. It would also make your board, baught on Tindie, a drop in component, without the need to buy extra stuff elsewhere. 
It woul be nice to just put this board in between the existing connection on the stove controller.

Flobul, 2024-03-27:
Some stoves have a spare RJ11 connector on the back of it.
I wonder, what is the rate of stoves without/with spare RJ11 on the back ?

I think, adding a RJ11 to the board would make it bigger and the board would not fit the size of the ESP8266 Wemos anymore.

Domochip, 2024-09-04:
I agree with @Flobul, there is not so much stove without the rj11 on the back.
Anyway, I documented the Y adapter required and also, I have some in stock too.

closed 2024-09-04 by Domochip