Domochip/WPalaControl #39: No stove connection during initialization byte 0x201c in iGetStatusAtech function

dxmen7, 2024-08-10:
Hello, I have following issue, I have debugged every single line until Palazzetti.cpp line 1013:
Palazzetti::CommandResult Palazzetti::iGetStatusAtech()
{
    CommandResult cmdRes;
    // die zeile gibt nicht erwarteten wert zurück
    cmdRes = fumisComReadByte(0x201C, &_STATUS);
    if (cmdRes != CommandResult::OK) {
        Serial1.println("fumisComReadByte(0x201C, &_STATUS) != CommandResult::OK");
        return cmdRes;
    }

Here the controller tries to read byte =x201c and expects specifc byte value, but this failed. Without knowledge about Fumis byte structure it's difficult for me to go ahead.
DO you have an idea or more background knowledge about Fumis byte structure?

I have a Fumis Alpha UI from JOTUL stove connected to Fumis controller, where I connected my RJ11 cable to my controller.
If you need further information, please let me know.

Debugged serial Rx/Tx on stove and µC side looks fine.

Domochip, 2024-08-30:
Hello,

What is your motherboard model?

Domochip, 2024-09-04:
I can only imagine 2 scenario : 
 - there is a problem with the adapter and/or cable
 - your  motherboard is not compatible

Could you give me more details about your motherboard and explain how you cabled it?

dxmen7, 2024-09-04:
Hello,Thanks for your reply. Yes I will give you my motherboard model number this weekend, I have to partially dismount my stove. I checked the adapter several times with a friend together and measured every single element. In my opinion, there can be a risk with a newer hardware version from Fumis and potentially an incompatibility.

dxmen7, 2024-09-04:
Sorry,Forgot the second part. Cabled it by RJ11 cable and modified as you described in your repo with crossing the lines.This cable goes to adapter, adapter is on top of D1 Mini and d1 mini Ports also correct.With serial monitor I have checked Rx and Tx on both sides and can see an Tx by D1 Mini after receiving an Rx and vice versa from stove side.

Domochip, 2024-09-05:
ok, in that case I advise to take some pictures too at the same time you open it ;-)

dxmen7, 2024-09-08:
Hello from Germany,

I was not able to open side panel for direct access to motherboard.
I only have model type, vendor serial number under device model of UI panel at the side of the stove. 
Maybe this help you too.
Attached pictures how is the controller cabled to stove.
I will try gonget more information and manual, how to open side panel.

Please find here iCloud Link to pictures, attaching to mail is too big: https://share.icloud.com/photos/012Ni27n0eGaETrfiNwn6vIIw

Best,
Dennis

Domochip, 2024-09-08:
Hello,

When you plug the stove to the module without powering the d1mini, do you see the D1 diode blinking (activity of the stove bus)?
does an error appears on the stove screen after 10sec?
Same questions but with the d1mini powered?

dxmen7, 2024-09-08:
Hello,
Yes, LED on stove side is blinking and confirmed with bus analyzer, that Tx and Rx are transmitted and received frames.
D1 mini powered also.
NO error on stove side after 10 sec.

Domochip, 2024-09-08:
What's your Jotul stove model? Do you know if Jotul offers a "ConnectionBox" or a module to control it via mobile phone?

dxmen7, 2024-09-08:
It’s Model PF733 from Jotul.
Yes they offer a WiFi connection box with mobile app for one 200€, that was my trigger to try it by myself with your repo.

Service man also have a RJ11 cable with him to control stove via proprietary software provided by Jotul.
Connection box link with Jotul App you can find here: https://intl.jotul.com/sites/importers/files/products/JOTUL_PELLET_CONTROL_Manual_EN.pdf

Stove manual here: https://www.jotul.de/sites/deutsch/files/products/DE%20%20PF733-%20PNMU00049.pdf

Domochip, 2024-09-08:
OK, it should definitely be compatible with my module.

few seconds after powering the d1mini, you should see D3 led blinking shortly. Is that the case?

What's appears in the web page?

dxmen7, 2024-09-08:
Yes, D3 LED is blinking shortly after powering up.
Web Page looks like this:
￼
Uptime currently less than 1 min. But also after 5 mins same behavior.

Domochip, 2024-09-08:
The webpage image does not appears. could you paste it in github page directly pls
Which firmware are you using?

dxmen7, 2024-09-08:
[image]
 I am using firmware 2.1.5/3.1.2

Domochip, 2024-09-08:
OK, log is going out of D4 pin at 38400bps
First you can have look at what's logged.

Do you know how to create a platformio/arduinoIDE project and compile it?
if yes, I advidse you to use this very basic example : https://github.com/Domochip/Palazzetti/blob/master/examples/Serial_esp8266/Serial_esp8266.ino
if No, I can compile it for you

dxmen7, 2024-09-08:
Yes I have compiled it with Arduino IDE, but don't get any useful output. tried different Baudrates for serial monitor

Domochip, 2024-09-08:
What output do you get?

Domochip, 2024-09-08:
baudrate is 38400bps

Domochip, 2024-09-08:
the log is coming out of the D4 pin.
you need an additionnal serial adapter connected to GND and D4 to receive the log

dxmen7, 2024-09-08:
yeah was my fault, I will get one and come back to you

Domochip, 2024-09-08:
no problem ;-)

dxmen7, 2024-09-08:
Got it, unfortunately everything failed :(

wSer started
initialize Failed
myPala.getSetPoint failed
myPala.getAllTemps failed
myPala.GetStatus failed

Domochip, 2024-09-08:
do you have a logic analyzer?

dxmen7, 2024-09-08:
A friend of mine yes. What do you need for information?

Domochip, 2024-09-08:
The goal is to monitor exchanges on tx AND rx of the d1mini with real timing.
That's the ultimate solution to debug what's happening.

Another idea would be to log serial from micro-usb port + another adapter on D7.
But it needs to log the precise timing to be able to reconstruct the "discussion"

Domochip, 2024-09-11:
Hello, could you also try to remove R8 resistor or D3 led?

Domochip, 2024-09-11:
also what's your si8621 exact model (2 letters after 8621)?

dxmen7, 2024-09-11:
I have SI8621EC chip. Asked you former time as I ordered these. 
One additional hint, I am using white LEDs, tried red ones earlier, but then D1 Mini didn‘t Boot with Adapter because of too high current flow through LED, with white one and slight higher voltage drop, adapter boots fine.

Domochip, 2024-09-12:
thx, I looking for increasing the R8 value to 2Kohms instead of 1K (maybe more will be needed)
Another person reported it in issue #42

dxmen7, 2024-09-14:
Desoldered R8, nothing changes, what do you need exactly for Rx/Tx or what should I try?

dxmen7, 2024-09-14:
I did it and can provide a picture: D0 is RX, D2 is TX. After Reset of D1 mini, Snapshot was created from UART Decoder
[image]

dxmen7, 2024-09-14:
can also provide PulseView file: 
[dxmen7_domo_snapshot_rx_tx_pulseview.zip](https://github.com/user-attachments/files/17002379/dxmen7_domo_snapshot_rx_tx_pulseview.zip)

Domochip, 2024-09-14:
ok, this is exactly what I needed (pulseview file). I will compare it with mine shortly.
First look shows correct behavior but I need to check the answer of the stove.
when stove send a frame starting by 0x00, board can send a frame starting by 0x02 to read data. Stove immediately reply with a frame starting by 0x02 containing the answer

Domochip, 2024-09-18:
Hello, I just had mine connected and compared traces.
Yours:
[image]
Mine:
[image]

So, I can tell that you need to check this part:
[image]
Normally, when uC send something, it should be received back and in you case, nothing come back.

dxmen7, 2024-09-18:
Thanks for explanation! I have an idea. 

where does Pin 6 on controller side at RJ12 connector get the power of stove side in your description?
I see in my setup a big mistake, I have a RJ11 cable with RJ11 connector. Thought it was a RJ12 one with 6 wires. 

so in theory base of transistor Q1 don‘t get any power to connect. 

did you use normal phone cable RJ12 or RJ12 cross over or do it by yourself by cutting and soldering the lines one by one?

Domochip, 2024-09-18:
sorry, I didn't exactly understand what you mean.
As described here : https://github.com/Domochip/WirelessPalaControl#connect
I'm using an RJ11 cable (4 wires) and they are crossed (don't seems to be common in the end)

dxmen7, 2024-09-18:
Okay understood. I have also a RJ11 cable and crossed the wires by myself because didn‘t get online a cross over cable. 
with only 4 wires, I don‘t understand, how base of Q2 is powered of Pin 6 by J1?
In your picture, it looks like that out of J1 you have 5 lines, but only 4 are connected to the stove. 
when Q2 isn‘t switched up, I think there is the communication problem on my side at the uC.

dxmen7, 2024-09-18:
Ah, now got it.

Domochip, 2024-09-18:
Base of Q2 is powered by R5 or not powered when Q1 is ON
Pin 6 is not used and I don't know its real usage but there was a switch on my original CBox, so I've made this solder jumper for the future if we discover something new.

szengerle, 2024-09-21:
Supported @dxmen7 on hardware level - TL;DR: Q2 was the root-cause
With the knowledge that signal must be looped back to MCU / MCU TX must be on TxRx put the SuT into some test circuit:
- TxRx to high (1MΩ)
- Square TX signal from MCU (signal generator)
- VCC/GND on both sides 

Results:
- Q1 is working - feeding square into Q2's base
- Q2's collector is high - no square; 0/5V square on base
 
[image]
[image]

- 5V Emitter to VCC voltage

[image]

Replaced Q2:
- Voltage is droping now on base - square also on collector
[image]
[image]

- MCU side Rx shows also looped-back signal
[image]

@dxmen7 will test later with controller connected

dxmen7, 2024-09-21:
Now communication with stove works, as you said, after 10sec appears communication error on Stove screen with error code E004, is this the correct behaviour?

dxmen7, 2024-09-22:
Solved the last issue, cable was interrupted... now it works!
Thank you so much for your support! How can I donate your work?

Domochip, 2024-09-22:
Cool that it works now 👍 
I have a paypal at domochip@hotmail.com

closed 2024-09-22 by dxmen7

Domochip, 2024-09-22:
Thank you so much for your donation!
This helps me a lot for my home automation and electronics projects.
👍