Domochip/WPalaControl #41: Stove communication failed

1300371, 2024-09-08:
I use WPalaControl since 3 years now without issues.

Usually, i unplugged (from electric power) the stove and WPala module from winter to automn. 
I replugged recently my setup and saw that WPalaControl failed to communicate with stove.

My tries until then:
- check/switch cable
- update firmware from 1.4.1 to 2.2.2

When i plugged the stove to the module, some LED lights up but WPalaControl indicates:
`MSG : Stove communication failed! please check cabling to your stove.`

I changed nothing from last usage...

Is there a way to check logs from WPalaControl ? It's like the module can communicate with stove but info from stove are kind of wrong.

Domochip, 2024-09-08:
If you connect the module to the stove without powering it (micro-usb disconnected), do you see the D1 led blinking (next to the RJ11 connector)?

1300371, 2024-09-08:
Yes

Domochip, 2024-09-08:
The log is going out to the pin D4 on d1_mini. That means you need to solder a cable to a separate serial adapter.

Domochip, 2024-09-08:
otherwise, if you have an ESP32 miniKit like this : 
[image]
Then you can get the log directly to the usb serial of the module

1300371, 2024-09-08:
I use this

[image]
[image]
[image]
[image]
[image]

Domochip, 2024-09-08:
I advise you to check soldering of U1 maybe for a bad contact.
It is usually the difficult point where I'm finally cheking with microscope

1300371, 2024-09-08:
[image]

Maybe the issue ... Will try to resolder 🤞

1300371, 2024-09-08:
Tricky to sold ... but i was that !!
[image]
Thank you so much ❤️

closed 2024-09-08 by 1300371

Domochip, 2024-09-08:
yw 😉