Domochip/WPalaControl #42: ESP not booting with Adapter attached

Padm59, 2024-09-11:
When the adapter is connected with the Wemos D1 Mini, it will not boot because pin D8 is pulled high through R8 and LED3. 

As described in the Pinout below -> Best Pins to use - ESP8266:
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

Quick Fix: 
Just desolder R8 and it works.

Domochip, 2024-09-11:
Are you using a RED led too?

Padm59, 2024-09-11:
yes👍

Domochip, 2024-09-11:
what's your si8621 exact model? (two letters after 8621)

Padm59, 2024-09-11:
I have the si8621EC and besides of the following the exact same components.
I still have another connection problem besides the boot problem. This could be my fault though because I used another diode and not the SS34. Mine has a slightly higher forward voltage drop and is probably a bit slower switching.
I do get values and I have control half of the time. It loses the connection every other minute or so. When it looses the connection it only shows an empty serial number.

[image]
[image]

Domochip, 2024-09-22:
Hey, 
Did you made some progress? Do you need help?

Padm59, 2024-09-27:
Going to look into it again this weekend, maybe I get my pi pico to work as a logic analyzer.

Padm59, 2024-09-27:
I used the same red LED on the VDD input of the stove, maybe it just has a to high of a voltage drop to work correctly. When connected with the stove, the LED is also flickering all the time. For the power, I guess this could be the problem?
Also doing some sampling of rx and tx on the esp side. Will take a look at them later in detail.

I could try to just remove the LED D1 so that the level shifter and pullups have a higher voltage. tx and rx on the stove side is 5V so why should they be only pulled to roughly 3V or so.

Padm59, 2024-09-27:
[betweenR2andlevelshifter_txstove.zip](https://github.com/user-attachments/files/17169056/betweenR2andlevelshifter_txstove.zip)
[txrxesp_onpagerefresh_stoveconnected_withdata.zip](https://github.com/user-attachments/files/17169057/txrxesp_onpagerefresh_stoveconnected_withdata.zip)

This was recorded the moment after I refresh the webpage and I triggered of the tx channel. I had connection to the stove and also got data back in the web interface after the refresh. Maybe it helps, baudrate was 38400 right?
Also the tx line on the stove side looks really messy, probably because of the very destorted and low VDD line.

Domochip, 2024-09-27:
2 things: 
- regarding the schematic, I just reproduced the original one I had, so I don't know why it is powered this way (but is working fine)
- did you changed your diode to a proper SS34?

Padm59, 2024-09-27:
I still have the other diode on the board, but I bridged the LED, and now it's working perfectly fine so far.
I had a glance at the signal, and it looked much better without all the noise.
https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/3694/B1911USD-20D000114U1930.pdf
That's the red led I removed. So I have no led on the board at the moment. 
Doing a longer test tonight to see if there are still any communication errors. We will see tomorrow.

Domochip, 2024-09-28:
I'm using any red led found on aliexpress. I can only advise you to use an SS34 as specified in the schematic...

Padm59, 2024-10-02:
Got my self a SS34 yesterday, and now it works flawlessly. I did put the Red Led on VCC back and also have no noise anymore.
Now we know for sure that you won't get away with just unsing another diode😬
Thank you for your help and advise on this one

Back to the original reason why I opend this issue:

When either the LED3 or R8 are removed, it works. I don't need the led anyways. 
Maybe use for R8 2kOhms or higher this could also work.

closed 2024-10-02 by Padm59