Domochip/WPalaControl #43: Aide: Home Assistant Discovery

Gamodeur, 2024-09-11:
Bonjour,

Merci beaucoup pour votre excellent projet.
De mon côté, j’ai effectué toute l’installation et tout fonctionne bien, sauf le « Home Assistant Discovery ».

Si j’ai bien compris, le « Home Assistant Discovery » permet de remonter automatiquement les entités dans Home Assistant ?

En revanche, j’arrive à bien configurer une entité en l’ajoutant manuellement dans le fichier de configuration.

Est-ce que l’option « Base Topic » a un impact sur ce paramètre ?

(Build version : 2.2.2/3.1.3)

Ma config: 
[image]

Version Home Assistant:
[image]

Version Mosquitto broker: 6.4.1

Domochip, 2024-09-11:
Première question : Est-ce que le module se connecte bien au poele et au broker MQTT?
(le numéro de série apparait et l'état est Connected)
[image]

Pour répondre aussi à votre question : Est-ce que l’option « Base Topic » a un impact sur ce paramètre ?
Non, cela n'a pas d'impact, les données d'autodiscovery pour Home Assistant sont mise à jour si cette valeur change.

Gamodeur, 2024-09-11:
J'ai pas de numéro de série par contre.
[image]

[image]

Donner visible avec MQTT Explorer
[image]

Domochip, 2024-09-11:
OK, la partie qui va m'intéresser maintenant, c'est ce qu'il y a dans les topics sous homeassistant : 
[image]
Les premières entités publiées sont celles-ci. L'une correspond à l'état de connexion du module WPalaControl et la seconde à l'état de connexion du poêle. Sont-elles présentes?

Gamodeur, 2024-09-11:
J'ai rien qui correspond au module WPalaControl

Domochip, 2024-09-12:
ok, je vois.
Je vérifie que le numéro de série est valide avant d'envoyer l'autodiscovery.
Et là, le numéro de série est 00...00
Je vais enlever ce contrôle.
Par curiosité, de quelle marque et modèle de poêle s'agit-il?

Domochip, 2024-09-12:
Chez Palazzetti, le numéro de série commence par LT[année de fabrication} puis des chiffres.
Est-ce que le numéro de série JOTUL suit ce format?
Je pense surtout que JOTUL met les cartes Fumis dans les poeles et ne programme pas le numéro de série ensuite, tout simplement 😄

Gamodeur, 2024-09-12:
Oui mon numéro de série commence par LT2015
Après sur l'étiquette arrière il nomme ça "Matr N." et non "SN"

Oui je pense aussi et ça m'étonnerait pas 😄

closed 2024-09-12 by Domochip