Domochip/WPalaControl #44: Version zigbee

djpostka, 2024-09-19:
Bonjour, 
Serait-ce possible d'envisager une version zigbee? peut-être a base de ESP32-H2.

Je n'ai pas conscience du travail que cela demande mais je suis prêt à aider si nécessaire. Merci

Domochip, 2024-09-19:
Bonjour,

Je voyais plus cela à base d'ESP32-C6 (format SuperMini) mais après avoir jeter un œil, le travail nécessaires est très important.
Il fut d'abord se plonger dans le fonctionnement de Zigbee pour déclarer les bons endpoint/clusters, traiter les messages, etc. tout ça en ESP-IDF car Arduino n'est pas complètement porté
Et une fois implémenté, il faut faire intégrer le module dans la base de zigbee2mqtt.
Je voulais me faire la main en adaptant mon projet WDS18B20 en ZDS18B20, mais il faut beaucoup de temps que je n'ai pas pour le moment.

Domochip, 2024-09-19:
Autre point, si on regroupe ensemble toutes les infos disponibles, on doit parler de plus d'une centaine de sensor déjà (rien que le GET+STDT en ramène une cinquantaine).
Je n'imagine donc pas le nombre d'input qu'il faudrait pour être capable de faire l'équivalent des commandes

djpostka, 2024-09-19:
Ah oui je ne connaissais pas le ESP32-C6, et je ne pensais pas qu'il y aurais tant de choses à faire... Bon, mon poêle est largement à portée du wifi donc ce n'est pas la mort, mais comme j'essaye de passé toutes ma domotique en zigbee, je me disait que ce serait pas mal. Je croise les doigts pour qu'un jour tu est le temps et la motivation de le faire ^^. 
Merci a toi d'avoir pris le temps de répondre.

Domochip, 2024-09-19:
de rien ;-)

closed 2024-09-20 by Domochip