Domochip/WPalaControl #45: Découvert impossible dans Home Assistant

Archange2222, 2024-09-22:
Bonjour

J'ai correctement configuré l'ESP et la connexion avec Home Status est OK

> Home Automation Status
> MQTT Connection State : Connected
> Last Publish Result : OK

Cependant Home Assistant ne le détecte pas automatiquement : 

Mosquitto broker : 6.4.1 
Home Assistant : 
> Core 2024.9.2
> Supervisor 2024.09.1
> Operating System 13.1
> Interface utilisateur 20240909.1

Merci

[image]

Domochip, 2024-09-22:
Bonjour,

Pourriez-vous me confirmer que le numéro de série du Poêle apparait sur la page de status (même si ce ne sont que des 0 pour Jotul): 
[image]

Il faut ensuite utiliser "MQTT Explorer" et jeter un œil à ces topics (sont-ils présents):
[image]

Bonne Soirée

Archange2222, 2024-09-22:
Bonsoir 

Le numero de serie apparaît bien 

[image]

Dans MQTT explorer j'ai cela :

[image]

Domochip, 2024-09-23:
Bonjour,

Dans MQTT Explorer, c'est la partie dans **homeassistant** qui m'intéresse

Domochip, 2024-09-24:
Bonsoir,

Votre problème est-il résolu? Sinon, pourriez-vous me transmettre une capture de la partie **homeassistant** dans MQTT Explorer comme sur ma capture : 
[image]

Archange2222, 2024-09-24:
Bonsoir, 

pardon, voilà le bon imprim écran 

[image]

Domochip, 2024-09-24:
ok, le module envoie donc correctement les infos à homeassistant.
Est-ce que les appareils sont bien apparus dans Home Assistant comme ceci (WPalaControl et Stove) : 
[image]

Archange2222, 2024-09-25:
Bonjour 

c’est bon il apparaît bien correctement, j’étais persuadé qu’il fallait faire un ajout manuel comme un nouvel appareil. Mais c’est bon il remonte bien et j’ai bien toutes les informations. 

Merci pour votre aide et bonne journée

closed 2024-09-25 by Archange2222