Domochip/WPalaControl #47: Ventilation arrière

ec0li82, 2024-09-24:
Bonjour
Je viens de recevoir mon module, cela fonctionne, merci.
Sur mon poele (Palazetti Julie), j'ai une sortie arrière pour chauffer une pièce annexe.

J'ai une entité "left fan" qui a été détectée mais ne fonctionne pas (rien ne se passe, enfin si, le log book montre une déconnexion!).
[image]

J'ai remarqué que mon ventilateur arrière correspond à F4L :

[image]

Comment faire utiliser cette valeur dans home assistant?

Merci.

Domochip, 2024-09-24:
Bonjour,

Pourriez-vous m'indiquer le modèle de poêle que vous possédez?
Afin de comprendre cette inversion, pourriez-vous me transmettre les résultats des 2 pages web suivantes: 
http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=GET+STDT
http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=GET+ALLS

Merci

ec0li82, 2024-09-24:
Les voici
{"INFO":{"CMD":"GET STDT","RSP":"OK"},"DATA":{"LABEL":"WirelessPalaControl","GWDEVICE":"wlan0","MAC":"34:94:54:8E:28:F3","GATEWAY":"192.168.0.1","DNS":["192.168.0.1"],"WMAC":"34:94:54:8E:28:F3","WMODE":"sta","WADR":"192.168.0.25","WGW":"192.168.0.1","WENC":"psk2","WPWR":"-59 dBm","WSSID":"BLAISE","WPR":"dhcp","WMSK":"255.255.255.0","WBCST":"192.168.0.255","WCH":"11","EPR":"dhcp","EGW":"0.0.0.0","EMSK":"0.0.0.0","EADR":"0.0.0.0","EMAC":"34:94:54:8E:28:F3","ECBL":"down","EBCST":"","APLCONN":1,"ICONN":0,"CBTYPE":"miniembplug","sendmsg":"2.1.2 2018-03-28 10:19:09","plzbridge":"2.2.1 2022-10-24 11:13:21","SYSTEM":"2.5.3 2021-10-08 10:30:20 (657c8cf)","CLOUD_ENABLED":true,"SN":"000000000000000000000000000","SNCHK":0,"MBTYPE":0,"MOD":397,"VER":43,"CORE":130,"FWDATE":"2016-04-18","FLUID":0,"SPLMIN":6,"SPLMAX":51,"UICONFIG":2,"HWTYPE":6,"DSPTYPE":1,"DSPFWVER":29,"CONFIG":2,"PELLETTYPE":2,"PSENSTYPE":0,"PSENSLMAX":0,"PSENSLTSH":0,"PSENSLMIN":0,"MAINTPROBE":4,"STOVETYPE":1,"FAN2TYPE":3,"FAN2MODE":3,"BLEMBMODE":3,"BLEDSPMODE":1,"CHRONOTYPE":0,"AUTONOMYTYPE":1,"NOMINALPWR":9},"SUCCESS":true}

{"INFO":{"CMD":"GET ALLS","RSP":"OK"},"DATA":{"MBTYPE":0,"MAC":"34:94:54:8E:28:F3","MOD":397,"VER":43,"CORE":130,"FWDATE":"2016-04-18","APLTS":"2000-04-07 11:56:13","APLWDAY":7,"CHRSTATUS":0,"STATUS":0,"LSTATUS":0,"SETP":0.00,"PUMP":0,"PQT":12147,"F1V":0,"F1RPM":0,"F2L":2,"F2LF":0,"FANLMINMAX":[0,5,0,5,0,5],"F2V":0,"F3L":0,"F4L":3,"PWR":1,"FDR":0.00,"DPT":0,"DP":67,"IN":13,"OUT":0,"T1":0.00,"T2":20.70,"T3":24.00,"T4":0.00,"T5":20.10,"EFLAGS":0},"SUCCESS":true}

Merci.

Domochip, 2024-09-24:
Bonsoir,

Je viens de faire l'ajustement sur la branche master.
Je vous propose d'essayer cette version du firmware qui doit résoudre votre problème.
[v2.2.2-3.1.5.rc.zip](https://github.com/user-attachments/files/17121453/v2.2.2-3.1.5.rc.zip)

Bonne Soirée

ec0li82, 2024-09-25:
Merci beaucoup ça fonctionne! 
J'ai désormais également le setpoint. Celui ci reste a l'état inconnu. Peut être est ce normal si le poêle est éteint ?
Un get fonctionne: 
{"INFO":{"CMD":"GET SETP","RSP":"OK"},"DATA":{"SETP":23.00},"SUCCESS":true}

Domochip, 2024-09-25:
Pourriez-vous me donner le contenu de ces topics avec "MQTT Explorer" : 
WPalaControl/connected
homeassistant/number/WPalaControl_LT201XXXX_SETP/config
WPalaControl/SETP (attendre 1 minute max pour qu'il arrive)

ec0li82, 2024-09-25:
Bonjour, les voici.
Screenshot pour le 1er et le 3e:

[image]

Valeur pour le second :

`{"~":"WPalaControl","availability":{"topic":"~/connected","value_template":"{{ iif(int(value) > 0, 'online', 'offline') }}"},"command_template":"SET+SETP+{{ value }}","command_topic":"~/cmd","device":{"configuration_url":"http://wpalacontrol.local","identifiers":["WPalaControl_000000000000000000000000000"],"model":"397","name":"Stove","sw_version":"43 (2016-04-18)","via_device":"WPalaControl_3494548E28F3"},"device_class":"temperature","min":17,"max":23,"mode":"slider","name":"SetPoint","object_id":"stove_setp","state_topic":"~/SET","unique_id":"WPalaControl_000000000000000000000000000_SETP","unit_of_measurement":"°C","value_template":"{{ value_json.SETP }}"}`

Bizarre que je n'ai pas de numéro de serie LT alors que c'est un Palazzeti (après avoir lu d'autres issues, ca devrait etre le cas)?
Mon installateur n'a jamais su me donner de numéro de série quand je lui ai demandé, car Palazetti m'en demandait pour remplacer ma cbox HS.
Je n'ai aucune plaque ni aucun numéro inscrit à l'arrière...

Domochip, 2024-09-25:
ok, coté MQTT, tout est correct. L'entité SetPoint devrait être disponible. Je conseille juste un reboot du WPalaControl pour que ce soit repoussé. Si ça ne fonctionne toujours pas après, il faudra donc que l'on regarde coté HomeAssistant.

Très étrange oui pour le numéro de série. Ce sont bien des 0 pour les Jotul mais pour les poêles Palazzetti, il est sensé être programmé et étiqueté sur le poêle...

ec0li82, 2024-09-25:
OK. j'ai rebooté le wpalacontrol mais meme chose.
En fait l'entité est bien là mais la valeur n'est pas rapportée.
Par exemple via wpalacontrol je vois bien le setp actuellement à 21 (valeur egalement sur le poele)
Par  contre HA lui n'a pas de valeur. 
Sur la page web, le slider est au milieu (20 quand je passe avec la souris comme ici) :
[image]

En cliquant pour voir les details/l'historique, il ne recoit en fait aucune valeur
[image]

Si je met une valeur en utilisant le slider, le setp est bien appliqué dans wpala et constaté sur le poele. Cependant HA ne l'affiche toujours pas et le slider revient au centre après un refresh.
Si je change la temperature sur le poele, meme chose : le setp est bien recu sur wpala mais pas dans HA.

Je precise que le poele est sous tension mais pas actif, c'est peut etre normal tant qu'il n'est pas "allumé" ?

Domochip, 2024-09-25:
Merci beaucoup pour ces retours précis.
La consigne (aka setPoint) doit remonter même si le poêle est sur OFF.
Vos retours m'ont permis de trouver une erreur dans le code pour la découverte Home Assistant.
Voici donc la version corrigée : 
[v2.2.2-3.1.5.rc2.zip](https://github.com/user-attachments/files/17130229/v2.2.2-3.1.5.rc2.zip)

ec0li82, 2024-09-25:
C'est parfait avec cette nouvelle version !
Merci beaucoup !

Domochip, 2024-09-25:
Elle corrige ce problème, mais je viens d'en remarquer d'autres.
La valeur ne se met pas à jour immédiatement.
Je vais améliorer le fonctionnement avec ce mode de publication MQTT

Domochip, 2024-09-28:
Je viens de de faire une pré-release qui améliore tout cela du coup : https://github.com/Domochip/WirelessPalaControl/releases/tag/v2.3.1-3.1.5

ec0li82, 2024-09-28:
Effectivement c'est beaucoup plus réactif !
Merci!

Domochip, 2024-09-28:
;-) de rien

closed 2024-09-28 by Domochip