Domochip/WPalaControl #48: Failure to build related to ArduinoJson - regression

sven337, 2024-09-29:
Hi,
I am unable to build due to the following error:
```
src/base/Application.cpp: In member function 'bool Application::loadConfig()':
src/base/Application.cpp:105:18: error: 'ArduinoJson::V6214PB2::JsonDocument::JsonDocument()' is protected within this context
  105 |     JsonDocument jsonDoc;
      |                  ^~~~~~~
In file included from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.hpp:33,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.h:9,
                 from src/base/Application.h:14,
                 from src/base/Application.cpp:1:
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:280:3: note: declared protected here
  280 |   JsonDocument() : pool_(0, 0) {}
      |   ^~~~~~~~~~~~
```

This is a regression introduced by that commit:

commit 5165ec0d679e07a6e79a5fa38757287f62cfbb04
Author: DomoChip <domochip@hotmail.com>
Date:   Sat Feb 3 15:03:48 2024 +0100

    Update WirelessBase to 2.1.5

 src/base/Application.cpp | 42 ++++++++++++++----------------------------
 src/base/Application.h   |  2 +-
 src/base/Core.cpp        |  2 +-
 src/base/Core.h          |  2 +-
 src/base/Version.h       |  2 +-
 src/base/WifiMan.cpp     |  2 +-
 src/base/WifiMan.h       |  2 +-
 7 files changed, 20 insertions(+), 34 deletions(-)

sven337, 2024-09-29:
My apologies: I have little experience with platformio and did not know I was supposed to "platformio pkg update" first.
This builds fine with an up-to-date ArduinoJson.

closed 2024-09-29 by sven337

Domochip, 2024-09-29:
yes, I'm looking to define the minimum version of ArduinoJson in platformio.ini file