Domochip/WPalaControl #49: Home Assistant discovery: two devices are created

sven337, 2024-09-29:
Home Assistant discovery seems to create two devices for me.
My host name is "ControlePoele" and I can see a "ControlePoele" device which has the "Connectivity" entity and nothing else.
I can also see a "Stove" device which contains all entities (including "Connectivity").

Below is an exercpt of autodiscovery data. Do you have any idea what could be happening?
```
{
  "device_class": "connectivity",
  "device": {
    "configuration_url": "http://wpalacontrol.local",
    "identifiers": [
      "WPalaControl_LT201814180810080229554"
    ],
    "model": "338",
    "sw_version": "43 (2017-07-18)",
    "via_device": "WPalaControl_083A8DD44AD5",
    "name": "Stove"
  },
  "entity_category": "diagnostic",
  "object_id": "stove_connectivity",
  "state_topic": "controlepoele/connected",
  "unique_id": "WPalaControl_LT201814180810080229554_Connectivity",
  "value_template": "{{ iif(int(value) > 1, 'ON', 'OFF') }}",
  "platform": "mqtt"
}

{
  "device_class": "connectivity",
  "device": {
    "configuration_url": "http://wpalacontrol.local",
    "identifiers": [
      "WPalaControl_083A8DD44AD5"
    ],
    "manufacturer": "Domochip",
    "model": "WPalaControl",
    "sw_version": "v2.3.1-3.1.5",
    "name": "ControlePoele"
  },
  "entity_category": "diagnostic",
  "object_id": "wpalacontrol_connectivity",
  "state_topic": "controlepoele/connected",
  "unique_id": "WPalaControl_083A8DD44AD5_Connectivity",
  "value_template": "{{ iif(int(value) > 0, 'ON', 'OFF') }}",
  "platform": "mqtt"
}
```

Expected behavior would be to have everything under "ControlePoele" and manufacturer Domochip, where the "Stove" device doesn't have a manufacturer ID.

Thanks

Domochip, 2024-09-29:
This is normal you have 2 devices 😄 
This representation is following Home Assistant standard.
ControlePoele represent your WPalaControl module
Stove represent your stove

You can see that they have each their model and version.
And Stove is connected to Home Assistant via the WPalaControl module ("via_device" setting)

sven337, 2024-09-29:
Haha I didn't know this was how things were normally done.
Thanks for the explanation :smile:

closed 2024-09-29 by sven337

Domochip, 2024-09-29:
You're welcome