Domochip/WPalaControl #50: Not responsive after some time

ec0li82, 2024-10-07:
Hello
For the second time, my WirelessPalaControl became unresponsive. This time it's after 4 days. Latest firmware.
Not available/disconnected in ha
Web gui not responding
IP not pinging
RJ11 led stays red as normal
Only solution is replug USB power.

I'm wondering if it's DHCP (ip set in DHCP server)...I've set a static IP directly now...let's see.
Wondering about my USB power source, will change that next time...
How to troubleshoot further?
Thanks.

Domochip, 2024-10-08:
If there is no wifi connection, the module switch automatically to AccessPoint mode.
An interesting thing would be to go close to the module and see if the Wifi network named "WirelessPalaControlXXX" appears

ec0li82, 2024-10-08:
Happened again after 12 hours.
Was not in ap mode.
I've now changed the USB plug and cable. Let's see

ec0li82, 2024-10-14:
Seems stable now... Let's close

closed 2024-10-14 by ec0li82