Domochip/WPalaControl #51: cycle d allumage puis extinction

Rehcywgnaf, 2024-10-10:
Bonjour, quand j allumé mon poêle ( austroflamm ben)  via la commande ON/OFF sur home assistant, j ai un cycle de allumage qui se lance, le boutton On/Off devient indispo dans HA le poêle s allume puis s eteint dans la foulée. 2 précisions: le numero de série de mon poêle apparaît en 000000000 mais  la saison dernière cela n empêchait pas le module de commander le poêle.
Merci pour votre aide

Domochip, 2024-10-10:
Bonjour, 
Quelle version du firmware utilisez-vous?
Après cette opération, si vous vous rendez sur la page du module, quelle est la valeur de UpTime?
"le poêle s allume puis s eteint dans la foulée" : est-ce que l'on parle de 2 secondes, ou bien le feu est allumé puis le poele l'éteind?
Quelles sont les valeurs de consigne (SetPoint) et de temperature mesurée dans la pièce (Room Temperature)?

Rehcywgnaf, 2024-10-10:
Bonsoir le fw est en 3.1.8, sur la page du module je ne  trouve pas uptime je vois Ontime.
Le feu va bien jusqu'à l allumage, j ai du feu puis ça s eteint. c est comme si il faisait un cycle Fuelign puis firestop .

Concernant les consignes je vous joins une capture m. 
Merci 
[image]

Domochip, 2024-10-10:
Bonsoir,

l'UpTime se trouve juste en dessous de la version : 
[image]

J'ai l'impression que vous l'utilisez avec l'intégration de marcopal74 (https://github.com/marcopal74/home-assistant-palazzetti).
Est-ce exact?

Auriez-vous l'historique de l'entité WpalaControl? Je souhaiterais connaitre l'état du poêle suite à l'indisponibilité.

Cette année, j'ai rajouté la découverte automatique à Home Assistant via l'intégration MQTT.

Rehcywgnaf, 2024-10-12:
Bonjour, 
pardon pour le delai, 
c'est bien cette integration que j'utilise . 
Aujourd'hui nouveau test, j'ai des delais plus long mais quand meme un cycle qui revient sur FIRESTOP  sans avoir atteind la temperature cible.
[image]

[image]
[image]

Merci pour votre aide

Domochip, 2024-10-12:
Quel est l'historique de température ambiante?
Qu'y a-t-il dans le journal sur cette période aussi?

Rehcywgnaf, 2024-10-13:
Avec votre aide je pense avoir compris d'ou venait le probleme, j'ai un defaut sur la sonde de temprature de mon poele, du coup j' avais  configuré un "versatile thermostat" qui agissait sur le ON/OFF en se basant sur un capteur de temperature relié à mon HA . 
[image]

maintenant quand je desactive le thermostat, je n'ai plus le fameux cycle mais une montée en temperature importante.
est il possible de prendre comme reference une temperature autre que "TEMP. AMBIENTE" ? 
et enfin mon poele propose un mode "eco" quand la temperature est atteinte est il possible de le retrouver d'une certaine maniere dans le module ?

Domochip, 2024-10-13:
Bonsoir,

Je vous conseille d'abord de vérifier que la sonde de température est bien vissée au borne de la carte mère (T01).
Il n'est pas possible d'utiliser une autre température comme référence. C'est pour cela que j'ai développé le projet WPalaSensor.

Pour le mode éco: 
- Le paramètre 50 (PAR50) contrôle le démarrage du poêle
- Le paramètre 53 (PAR53) contrôle l'extinction du poêle

Voici les URL pour les obtenir (la valeur est à diviser par 10): 
http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=GET+PARM+50
http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=GET+PARM+53

Pour les modifier, il faut utiliser ces commandes (10 veut donc dire 1.0°C d'écart par rapport à la température de consigne):
http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=SET+PARM+50+10
http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=SET+PARM+53+10

Rehcywgnaf, 2024-10-14:
Bonjour, 

Merci pour ces réponses, je vois plus clair, je vais pouvoir cloturer ici. 

je n'avais pas du tout entendu parlé de WPalaSensor que je vais consulter de ce pas, je voulais contourner le remplacement d'une sonde proprietaire hors de prix. 
Je vais me pencher aussi sur les parametres du mode eco du Wpalacontrol. 

Merci je reviens vers vous en mp si besoin d'un pcb pour le Wpalasensor.

closed 2024-10-14 by Rehcywgnaf

Domochip, 2024-10-14:
Bonjour,

On peux trouver des sondes dans les 25€ sur le Net.

Bonne Journée