Domochip/WPalaControl #54: Stove communication failed

Palyndiel, 2024-10-29:
Bonjour,

Je n’arrive plus à faire fonctionner le module palacontrol avec mon poêle Turbo Fonte.
Le boîtier fonctionnait encore en début d’année mais en le rebranchant en début de mois, impossible de le faire fonctionner. 
J’ai le message d’erreur suivant :
[image]

Version du module : 
[image]

La petite led rouge est bien allumée : 
[image]

Merci de votre aide.

Domochip, 2024-10-29:
Je vous conseille de contacter le vendeur de votre module dans ce cas

Domochip, 2024-11-01:
Bonsoir,
Avez-vous eu un retour de votre vendeur?

arabine, 2024-11-02:
Bonjour, j'ai exactement le même problème avec un module que je viens d'acheter sur TIndie. Normalement j'ai le bon modèle de poële...

Domochip, 2024-11-02:
@arabine 
L'avez-vous acheté auprès de moi?
Si tel est le cas, pourriez-vous plutôt me contacter via Tindie. Je pense que c'est une meilleure option pour suivre le SAV

Bonne Journée

Domochip, 2024-11-02:
Petite clarification:
- Je ne sais pas qui est DIYPala (autre vendeur sur Tindie)
- J'ai ouvert ma boutique Tindie il y a un peu plus d'un mois sous mon pseudo Domochip
- Auparavant, les gens me contactaient par mail et je vendais en direct.
- @Palyndiel : je n'ai jamais fabriqué de PCB blanc. Je n'en suis donc pas le fabricant...
- @arabine : votre nom n'apparait pas dans ma liste de commande. Je ne pense donc pas en être le fabricant non plus.

Malgré tout, le but de ce projet est d'aider les gens à améliorer leur habitat.
Je n'ai pas de soucis à aider tout utilisateur mais pour des problèmes matériels comme les vôtres, il n'y a que 2 options:
- je suis le fabricant : j'assure le SAV
- je ne suis pas le fabricant : c'est à vous de chercher la panne avec mon aide

Si vous n'avez pas de retour de vos fabricant et que vous souhaiter résoudre le problème, n'hésitez pas à donner un maximum de détails (marque et modèle du poêle, année d'achat du WPalaControl, version actuelle du firmware, votre poêle a-t-il une prise accessible ou bien un splitter a dû être ajouté, etc, ...)

@arabine si vous entrez dans ce dernier cas, Il faudra plutôt que vous ouvriez votre propre issue

Bonne Journée

arabine, 2024-11-02:
@Domochip oui mon vendeur est bien DIYPala, je suis entré en contact avec lui il m'a répondu, c'est donc en cours. Si on ne parvient pas à trouver le problème j'ouvre un ticket ici, de tout tracer peut aider les prochaines personnes ayant le même soucis.

(j'ai du matos électronique et je code du firmware donc je peux aider).

Merci !!

Palyndiel, 2024-11-02:
Bonjour, petit retour de mon côté. Après avoir passé 1h à faire du debug dans platformio, et après plusieurs nettoyage/reecriture, le module est tombé en marche.
Donc le matériel fonctionne effectivement mais je pense qu’il y a bien un problème software.

Domochip, 2024-11-02:
Bonjour,
Avez-vous flashé le même firmware plusieurs fois?
Votre poêle est-il ancien? La version du firmware de votre carte Fumis est-elle ancienne? S'agit-il d'une Alpha 65?
Quelle modification software avez-vous faite afin que cela fonctionne?

Palyndiel, 2024-11-03:
Bonjour,
J'ai re-flashé plusieurs fois la version 3.1.11.
Le poêle est récent, modèle Turbo Fonte de 2020.
Aucune modification du firmware, j'ai simplement ajouté des logs pour tenter de debuguer.

J'ai maintenant un nouveau problème, le boitier m'indique qu'il est bien connecté au poêle et à MQTT, mais dans home assistant, la connexion ne cesse de sauté : 
[image]
En gris déconnecté et en jaune connecté. 
Du coup le pilotage via HA n'est pas possible. Peut être faut-il ouvrir un autre ticket pour ce problème.

Domochip, 2024-11-03:
Votre poêle a-t-il une prise RJ12 accessible ou bien un splitter a dû être ajouté?
Désolé, mais je persiste à croire à un problème matériel.
Que vous affiche la page web du module durant les périodes "Déconnecté"?

Palyndiel, 2024-11-03:
Oui je suis branché à une prise rj12 accessible directement à l’arrière de l’appareil. 
Le module m’affiche toujours connecté, même lorsque je suis déconnecté.

Domochip, 2024-11-03:
Connaissez-vous le modèle de la carte mère de votre poêle?

arabine, 2024-11-03:
Bonjour, j'ai réussi à réparer le module en suivant les conseils du vendeur : comme indiqué ci-dessus, j'ai repassé un coup de fer à souder sur toutes les pattes des composants et ça a fonctionné !

Voilà merci domochip en tout cas pour ta disponibilité et le projet.

Domochip, 2024-11-10:
Bonjour @Palyndiel 
J'ai relu ce ticket et j'avais omis un point intéressant: 
_Je n’arrive plus à faire fonctionner le module palacontrol avec mon poêle Turbo Fonte. Le boîtier fonctionnait encore en début d’année_

Concernant un problème software:
- de nombreux utilisateurs du WPalaControl ne rencontre pas ce soucis
- si cela fonctionnait bien en début d'année, pourriez-vous retester avec le firmware que vous aviez à ce moment là?
- afin de vérifier que la communication au poêle est opérationnelle, il faut l'interroger: envoyer une commande GET+TMPS par exemple

La source la plus probable du problème reste à mon sens un défaut de soudure sur le si8621.
On appelle ça une soudure sèche (ou aussi soudure collée). Il y a un infime espace entre la patte et la piste sur le PCB. Au début, il y a contact, puis avec le temps, il devient intermittent et finalement il n'y a plus de contact.
La difficulté vient du fait qu'en appuyant sur la patte du composant avec un multimètre pour vérifier la continuité, cela fait contact… puis quand on la relâche le problème revient…
C'est pour cela que je vérifie au microscope que c'est correctement soudé.

Bonne Journée

Palyndiel, 2024-11-12:
Bonjour, la connexion au poêle est stable depuis le 8 novembre. Plus de connexion/déconnexion intempestive. Sans aucune modification de ma part, juste en laissant le module branché.
Je vais néanmoins investir dans un kit de soudure, et essayer de ressouder le composant que vous m’indiquez si jamais je rencontre à nouveau des soucis.
En attendant je clôture le ticket pour ne pas vous embêter plus longtemps.
Merci du temps accordé.

closed 2024-11-12 by Palyndiel