Domochip/WPalaControl #58: SetPoint unknown in Home Assistant

tanc, 2024-11-08:
Hi, I just received my box this morning, nice easy set up, thanks!

The only issue I have is that the SETP isn't registering in Home Assistant and is showing as unknown. 
This is what MQTT Explorer says is showing:

[image]

```
{
  "~": "WPalaControl",
  "availability": {
    "topic": "~/connected",
    "value_template": "{{ iif(int(value) > 0, 'online', 'offline') }}"
  },
  "command_template": "SET+SETP+{{ value }}",
  "command_topic": "~/cmd",
  "device": {
    "configuration_url": "http://wpalacontrol.local",
    "identifiers": [
      "WPalaControl_LT201829880571372127806"
    ],
    "model": "83",
    "name": "Stove",
    "sw_version": "47 (2023-01-30)",
    "via_device": "WPalaControl_3494548E2B82"
  },
  "device_class": "temperature",
  "min": 17,
  "max": 23,
  "mode": "slider",
  "name": "SetPoint",
  "object_id": "stove_setp",
  "state_topic": "~/SETP",
  "unique_id": "WPalaControl_LT201829880571372127806_SETP",
  "unit_of_measurement": "°C",
  "value_template": "{{ value_json.SETP }}"
}
```

Looking at this could it be that the min/max are wrong? This stove is a hydro/boiler model (Ginger Idro 15) so the SetPoint is much higher as its setting the water temperature. Its min is 65 and max 80.

Also `RoomTemp` on this model should be `WaterTemp` with `"name": "Water Temperature",`

Domochip, 2024-11-08:
Hi,

Could you go to this page : 
[http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=GET+STDT](http://wpalacontrol.local/cgi-bin/sendmsg.lua?cmd=GET+STDT)
and give me the values of:
- FLUID
- SPLMIN
- SPLMAX

tanc, 2024-11-08:
```
"FLUID": 1,
"SPLMIN": 65,
"SPLMAX": 80,
```

Domochip, 2024-11-14:
Could you compile from the master branch and confirm it works fine for you, please

Domochip, 2024-11-15:
It's available in the last release

closed 2024-11-15 by Domochip