Domochip/WPalaControl #60: SetPoint min limité à 17 et Test Fire

arabine, 2024-11-12:
Bonjour,

Désolé d'avance si je suis hors sujet c'est peut-être une mauvaise configuration de mon poële. 

J'ai deux scripts dans HomeAssistant dont un qui place le SetPoint à 17. Première question : pourquoi je ne peux pas mettre moins ?

Ensuite, la nuit mon poële démarre tout seul, "test fire", je n'avais pas du tout ça avec la connection box :

[image]

Une idée du pourquoi ?

arabine, 2024-11-12:
Voici ma conf :

`{"INFO":{"CMD":"GET STDT","RSP":"OK"},"DATA":{"LABEL":"ESP-EF9672","GWDEVICE":"wlan0","MAC":"3C:61:05:EF:96:72","GATEWAY":"10.0.0.1","DNS":["10.0.0.1"],"WMAC":"3C:61:05:EF:96:72","WMODE":"sta","WADR":"10.0.0.6","WGW":"10.0.0.1","WENC":"psk2","WPWR":"-84 dBm","WSSID":"NETGEAR23","WPR":"dhcp","WMSK":"255.255.255.0","WBCST":"10.0.0.255","WCH":"11","EPR":"dhcp","EGW":"0.0.0.0","EMSK":"0.0.0.0","EADR":"0.0.0.0","EMAC":"3C:61:05:EF:96:72","ECBL":"down","EBCST":"","APLCONN":1,"ICONN":0,"CBTYPE":"miniembplug","sendmsg":"2.1.2 2018-03-28 10:19:09","plzbridge":"2.2.1 2022-10-24 11:13:21","SYSTEM":"2.5.3 2021-10-08 10:30:20 (657c8cf)","CLOUD_ENABLED":true,"SN":"LT201631680580338070098","SNCHK":1,"MBTYPE":0,"MOD":390,"VER":43,"CORE":129,"FWDATE":"2016-08-23","FLUID":0,"SPLMIN":6,"SPLMAX":51,"UICONFIG":1,"HWTYPE":6,"DSPTYPE":1,"DSPFWVER":29,"CONFIG":1,"PELLETTYPE":3,"PSENSTYPE":0,"PSENSLMAX":0,"PSENSLTSH":0,"PSENSLMIN":0,"MAINTPROBE":4,"STOVETYPE":1,"FAN2TYPE":4,"FAN2MODE":3,"BLEMBMODE":1,"BLEDSPMODE":1,"CHRONOTYPE":0,"AUTONOMYTYPE":2,"NOMINALPWR":12},"SUCCESS":true}`

Domochip, 2024-11-12:
Quelle serait votre valeur minimum?
Je vois que la nouvelle integration Palazzetti configure le climate pour aller de 6 à 51.

arabine, 2024-11-12:
Ok pour le test fire. J'ai un événement qui met le SetPoint à 16 le soir et ceci pour toute la nuit, peut-être qu'elle n'est pas acceptée par le module ou la limite de 17°C minimum est purement graphique ? Ma température de la pièce ne descend jamais en dessous de 18°C et pourtant le poële démarre.

Dans l'idéal, si on veut copier le fonctionnement d'un fil pilote :

- Confort : température de consigne en usage normal réglable de 7 à 30 °C par l'utilisateur
- Eco ou réduit : température de confort abaissée de 3,5°C
- Hors gel : régulation réglée sur 7 à 8°C

Il faudrait pouvoir descendre à au moins 12/13°C je pense (19°C préconisé pour les pièces de vie -3°C pour le mode Eco et encore -3 pour le mode hors gel).

Le max, au delà de 25°C ce n'est pas nécessaire je pense.

Pour le réglage du SetPoint je passerai plutôt par un popup/validation et éviter les miss-clicks avec le slider (on sait jamais trop à quel moment la commande est envoyée).

Domochip, 2024-11-12:
Je pense que la commande à 16 ne passe pas à cause de la limite : n'y a-t-il pas de message d'erreur dans les logs?

Concernant la limite basse, le mode Hors-Gel peut être ignoré, la carte mère possède déjà une fonction Hors-Gel nativement.
Ca donnerait plutôt 15°C

arabine, 2024-11-13:
yep 15°C serait pas mal, ça évite d'éteindre le poële la nuit (et une nuit bien froide d'hiver dans ma maison en pierre ça descend bien bas !!).

Cette nuit ça s'est bien passé, donc le SetPoint à 17 est correctement passé et effectivement le SetPoint à 16 pose des problème et c'est dommage de bloquer à ce niveau là.

Merci pour l'analyse.

Domochip, 2024-11-15:
C'est dans la dernière release

closed 2024-11-15 by Domochip

arabine, 2024-11-18:
Ok top merci ! (ça n'a pas pris de temps ce n'était pas un sujet urgent ne t'inquiète pas !!)

Domochip, 2024-11-18:
Une pre-release sortie hier soir change la donne aussi :-D
Elle propose une entité climate plus conviviale.
L'entité SetPoint d'avant reste toujours disponible mais les limites sont maintenant celles du poêle (souvent de 6 à 51°C pour les type Air).