Domochip/WPalaControl #61: Feature request: Expose the firebox temp to Home Assistant

tanc, 2024-11-15:
For my nerdy graph needs it would be great to expose the T3 temp to Home Assistant so that I can monitor the relationship between the tank temp and the firebox temp.

```
{
    "INFO": {
        "CMD": "GET TMPS",
        "RSP": "OK"
    },
    "DATA": {
        "T1": 71.9, <-- Tank or outflow temp
        "T2": 63.9, <-- Inflow temp
        "T3": 210, <-- Firebox temp
        "T4": 0,
        "T5": 18 <-- Room temp
    },
    "SUCCESS": true
}
``

Domochip, 2024-11-16:
It's added in the current master branch as a disabled by default one

BorisNanev, 2024-11-16:
"T3": 210, <-- Firebox temp is important for monitoring how stove works and when exhaust gases go too much UP is indicator that your stove is for cleaning (i mean inside of waste box).

tanc, 2024-11-16:
> It's added in the current master branch as a disabled by default one

Should I be able to see it and enable it in Home Assistant already? Or do you mean it needs to be enabled in code first?

BorisNanev, 2024-11-16:
i cant see it in HA. only in browser of domochip.

Domochip, 2024-11-17:
@BorisNanev It is in master branch not part of the current release. You need to compile and flash it in your module if you want to see it

@tanc it will be created in Home Assistant but you'll need to go to the device and enable this entity, then you'll be able to use it like any other

Domochip, 2024-11-17:
going to the stove device, you can find some grayed out sensors, and enable them : 
[image]
[image]
[image]

BorisNanev, 2024-11-17:
I know, I enabled all sensors in home assistant but I don't have T3. I can a see it in browser but not in hove assistant .

Domochip, 2024-11-17:
You'll find the version containing these new sensors in the latest pre-release (3.4.0)
https://github.com/Domochip/WPalaControl/releases

BorisNanev, 2024-11-17:
That's awesome. Thanks one more time. I was looking for this from years. :)

Domochip, 2024-11-19:
@tanc @BorisNanev Does everything looks good for you?
I'm considering switch the current pre-release as release

BorisNanev, 2024-11-19:
I don't have any issues with pre-release. Everything works perfectly. I do have some random disconnects for 2-3 seconds 3-4 time per day, same as stable release.(I think it's not from bad WiFi connection coz router and device show good connection). It's not big of deal for me.

Domochip, 2024-11-20:
@BorisNanev thank you for the feeback.
Regarding the disconnection, I have it sometime and sometime not (3 days without it on my side now...)
I was not able to catch it yet.
Could you confirm me this behavior : 
 - in Home assistant, both Stove and WPalaControl connectivity disconnect and reconnect
 - in WPalaControl webpage, the UpTime is not resetted

BorisNanev, 2024-11-20:
Yes, I can confirm. 35 hours without disconnection. I didn't check UpTime but I will next time.

Domochip, 2024-11-20:
ok, I'll close this issue.
Regarding the disconnection, it is still very hard to figure out now.
I'll have a look at this later when I've time.

closed 2024-11-20 by Domochip

tanc, 2024-11-21:
Sorry I wasn't able to test, but just wanted to say thanks, I did the update today and can confirm the fluegas sensor is now available (after enabling it)