Domochip/WPalaControl #63: Online updating not working for me

tanc, 2024-11-21:
For some reason new firmware isn't found and it doesn't allow me to update directly online, I need to use the manual update option. Clicking on the URL field in the online update section takes me to a github 404 page, I don't know if that is related to the issue.

[image]

[image]

Domochip, 2024-11-21:
Could you press F12 on your browser and then go to the firmware page.
What's interesting me is the content returned by /glui request
like this :

tanc, 2024-11-21:
```
{
    "installed_version": "3.4.0",
    "latest_version": "",
    "title": "",
    "release_summary": "",
    "release_url": "https://github.com/Domochip/WPalaControl/releases/tag/",
    "release_date": ""
}
```

Domochip, 2024-11-21:
ok, thank you, it helps me to target the problem

Domochip, 2024-11-23:
I think I found it.
Should be solved by ff07631a19eb26b847c5f0b5df96c576ac7896d0

Domochip, 2024-11-24:
it is included into the 3.5.0 pre-release

BorisNanev, 2024-11-26:
I do update it today and have same problem. Every update i do manualy.

Domochip, 2024-11-27:
@BorisNanev browser doesn't matter. Could you give me a screenshot of it?

BorisNanev, 2024-11-27:
Here, same as first post.
[image]

Domochip, 2024-11-27:
@BorisNanev 
ok, it is different than the tanc case.
There is no URL so no new version has been gathered by the module.
Which probably means that the module was not able to contact Github.
What's the IP configuration of your module?

BorisNanev, 2024-11-27:
There was a URL before update. with 3.5.0 i i dont see it. What you mean with IP configuration?

Domochip, 2024-11-27:
If you go to Config, did you ticked the Static IP box and configured an IP address? 
[image]

Domochip, 2024-11-27:
Strange, the problem is that I'm not able to reproduce it.
Is it empty everytime you go to firmware page?
Here is what I currently get: 
[image]

Domochip, 2024-11-27:
Could you try to go to this url and tell me how long it takes to load : 
https://api.github.com/repos/domochip/WPalaControl/releases/latest

BorisNanev, 2024-11-27:
[image]

I dont know if this help but...

BorisNanev, 2024-11-27:
> Could you try to go to this url and tell me how long it takes to load : https://api.github.com/repos/domochip/WPalaControl/releases/latest

Instantly. Abslute no delay

BorisNanev, 2024-11-27:
Can you downgrade the firmware?

Domochip, 2024-11-27:
yes

BorisNanev, 2024-11-29:
I downgraded and URL are showing up. Then I upgrade and no more URL.