Domochip/WPalaControl #66: Puissance Palazzetti Emily 9

hawaien88, 2025-01-03:
Bonjour,

Reçu aujourd'hui le module et mise en place ultra simplement. J'ai tout de même une question, j'ai un Palazzetti Emily 9 et dans la découverte, il ne m'a pas ajouter la puissance de 1 à 5. Dois-je faire quelque chose de plus?

[image]

Merci pour votre aide.

Domochip, 2025-01-03:
Bonjour,
Je comprend par votre capture que vous utilisez l'intégration Palazzetti.
Plusieurs fonctionnalités manquantes vont arriver avec Home Assistant 2025.1 (livraison prévues ce soir!)
C'est @dotvav qui le développe.
De mon coté, le module prévois une intégration native et plus complète en passant par l'intégration MQTT.

hawaien88, 2025-01-03:
Top pour la mise à jour.
Entre temps, je me suis quand même penché sur la compréhension de mqtt que je n'avais pas réussi à utiliser au départ. Après avoir cherché, j'ai découvert mon problème qui était dans les login et password MQTT.
Je vais pouvoir utilisé MQTT finalement.

Merci pour les infos et ce module au top !

[image]

closed 2025-01-03 by hawaien88

Domochip, 2025-01-03:
Parfait 👍 
Avec plaisir.
Bonne Soirée