Domochip/WPalaControl #67: Connexion filaire?

BEBAR-BID, 2025-01-20:
Bonjour Domochip,

Je dispose d'une prise RJ45 sur le mur, située derrière mon poêle, qui pourrait permettre de connecter ce dernier au boîtier Home Assistant via un serveur. Je tiens particulièrement à réaliser l'intégralité de mon installation domotique (notamment en utilisant le protocole KNX) en filaire.

Est-il possible de communiquer avec le module WpalaControle en filaire (sans utiliser le WiFi) ?
Si ce n'est pas le cas, comment adapter ce module pour permettre une communication via un câble RJ45 ?

L'utilisation d'un ESP32 ou d'un Arduino avec un module Ethernet (comme le W5500) est-elle ma seule alternative ?

Merci d'avance pour votre aide.

Domochip, 2025-01-21:
Bonjour,

Il n'est pas possible de communiquer avec le WPalaControl en filaire.
Pour l'adapter, je vous conseille donc de commencer par le projet https://github.com/Domochip/WBase
C'est le socle de base pour mes différents projets (c'est lui qui gère la configuration système dont la partie Wifi).

Ces projets sont compatibles ESP32 mais pas activement supportés sur cette plateforme.
Je ne sais pas si il existe d'autres alternatives hardware.

Bonne Journée

BEBAR-BID, 2025-01-21:
merci pour votre réponse

closed 2025-01-22 by Domochip