Domochip/WPalaControl #68: WPalaControl unstable

abcklaus, 2025-02-02:
Hi,

I am using the WPalaControl with Palazzetti "Marianne" stove. For any reason the WPalaControl seems to be unstable. 

The uptime in the status indicates that the device is restarting. In this example the uptime is about 2h. Actually the last power cycle is about 20h ago ...

[image]

In Home Assistant I am using the Palazzetti integration. Also here I am observing availability issues:

[image]

I other issues I have seen that the power supply may be a problem. But after trying three different power supplies I am pretty sure that this is not the root cause.

Any Idea what I can do?

Thanks

Michael

Domochip, 2025-02-03:
Hi,

This may happen if there is some Wifi connectivity issue.
What is the percentage next to your wifi network in Config page?

abcklaus, 2025-02-03:
It’s 58%. Distance between access point and Domochip is about 5m.

abcklaus, 2025-02-03:
ok, thank you. I didn't expect that the quality of the wifi connection may cause issues like that.

In the meantime I did some tweaks in my environment and will observe.

abcklaus, 2025-02-06:
I did some tweaks in my wifi environment. The quality is now between 90 and 100%

[image]

Today in the morning (07:40) I see that the uptime is again just 2h. Means that the device has restarted at about 05:00

[image]

Is the wifi connection still an issue?

Domochip, 2025-02-07:
Hi,

Before going more in technical research, is it possible for you to use the MQTT integration (native for this module)?
I want to clarify if that come from HTTP endpoints or not

abcklaus, 2025-02-07:
ok, no problem. I have deactivated the Palazzetti integration in HomeAssistant and activated the MQ integration.

abcklaus, 2025-02-11:
The MQTT Integration is active for some days now.

But there is no improvement. The uptime is still just some hours. The max value I have seen is 1d.

In HA the stove becomes unavailable from time to time.

Domochip, 2025-02-23:
Hi,

Sorry I was off these last two weeks.
mm, sounds like an hardware problem.
Which WPalaControl HW version do you have (1.1 or 2.0) (it's written on the box and on the PCB).
Did you build it yourself or bought it ready to use and where?

abcklaus, 2025-02-23:
Hi,

I am using Version 2.0.

Domochip, 2025-03-03:
Have you bought it from me or from DIYPala?

abcklaus, 2025-03-04:
I bought it from you. Order number is  #515307

Domochip, 2025-03-05:
ok
Do you see a kind of pattern of the restart?
Next steps would be to monitor via USB from a computer and flash some test firmware (going deeper then).
Is it something you are comfortable with? otherwise a return/refund is always possible.

abcklaus, 2025-03-06:
Hi,
yes, flashing some firmware is no problem. I am familiar with that.

I have here an Lolin D1 mini from another project. What do you think - is it worth to flash your firmware to check if the issue is located in the other D1 mini?

Domochip, 2025-03-06:
yes, can be a good first test.

abcklaus, 2025-03-06:
I used a Lolin D1 Mini v4.0. I have compiled and flashed the firmware with vscode/platform io. For and reason this device seems not to be compatible. As soon as i have connected my stove, the stove Shows the error E0004.
So I moved back to the original D1 mini.

abcklaus, 2025-03-07:
ok, found the reason why my D1 mini is not working. The pins are soldered on the other side of the board.

[image]

abcklaus, 2025-03-07:
ok, found a workaround ;-)

[image]

Let's see how the new D1 mini will behave ...

abcklaus, 2025-03-22:
During the last days I have monitored the device. Unfortunately switching the D1mini did not help. I am observing the instability. In most cases the device does not have an uptime larger than several hours.