Domochip/WPalaControl #69: Stove im Home Assistant

abcklaus, 2025-02-02:
Hi,

I am using the WPalaControl with Palazzetti "Marianne" stove. I am using WPalaControl with Palazzetti integration in Home Assistant. 

I have observed that the stove becomes unavailable in HA if the target temperature is change while the stove is powered off:

[image]

Is that behavior intended?

Thanks

Michael

Domochip, 2025-02-03:
Hi,

Do you mean the target temperature is changed directly on the stove physically or via HA?

abcklaus, 2025-02-03:
It happens when I change the target temp in HA.

Domochip, 2025-02-03:
Interesting, 
could you give it a try directly using HTTP command and check that it disconnect or not (ex: set target temperature to 20):
http://{WPalaControlIP}/cgi-bin/sendmsg.lua?cmd=SET+SETP+20

abcklaus, 2025-02-03:
Response for HTTP command:

{"INFO":{"CMD":"SET SETP","RSP":"OK"},"DATA":{"SETP":20.00},"SUCCESS":true}

One more thing: When setting the temp in HA I get this error:

[image]

abcklaus, 2025-02-03:
After executing the HTTP command WPalaControl remains available. But it takes some seconds (approximately 10s) to reflect the change in HA.

Ok, I will contact dotav.

Domochip, 2025-02-03:
For the 10sec delay, it is normal, Palazzetti integration is using HTTP endpoints and no notification is sent from WPalaControl nor CBox using this protocol.
What's interesting is that HA didn't considered the WPalaControl has disconnected, so yes, the problem is coming from Palazzetti integration

dotvav, 2025-02-03:
Hi there. It could be that the response payload from WPalaControl has a
format that I was not expecting. If the processing throws an exception
(missing attribute, unexpected type...) then the device will be reported as
unavailable, until it responds again with something that fulfills the
expectations. Which could be 10s. I haven't had a chance to test today, but
I have some doubts about SETP returning a float value instead of an int.
I'll check tomorrow.

Domochip, 2025-02-03:
Nice trail found. If my memory is good, some stove are able to be configured with 0.2°C step, but in that case, i don't know what CBox returns...

Domochip, 2025-03-05:
@dotvav were you able to confirm if it comes from the float returned?

dotvav, 2025-03-05:
Hi, I came back from vacation last week and haven't found the time to run these tests. I will look into that in the coming days.

dotvav, 2025-03-09:
I ran some tests today, both on the HA integration and on the API lib, and the `float` type of SETP does not cause any issues. 

@abcklaus, can you open an issue on the HA Core Github? It will get assigned to me. I don't want to polute Domochip's repo. Feel free to link to this issue and share the tests that you have performed. Also, any error message ort information you can get from the HA logs would be very helpful.

abcklaus, 2025-03-09:
ok, done.

#140215

Thanks

dotvav, 2025-03-10:
@Domochip below is the relevant error from @abcklaus's logs. There is a imeout error when I call `GET STDT`. Do you think that it is possible that the WPalaControl could struggle with concurrent http requests if HA is calling `GET STDT` right after (or during) the `SET SETP` command? (I'll also try to understand why I call `GET STDT` at this point)

````
2025-03-09 18:16:44.569 ERROR (MainThread) [homeassistant.components.palazzetti] Communication error
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/connector.py", line 1116, in _wrap_create_connection
    sock = await aiohappyeyeballs.start_connection(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/aiohappyeyeballs/impl.py", line 73, in start_connection
    sock = await _connect_sock(
           ^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/aiohappyeyeballs/impl.py", line 208, in _connect_sock
    await loop.sock_connect(sock, address)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 641, in sock_connect
    return await fut
           ^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client.py", line 703, in _request
    conn = await self._connector.connect(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        req, traces=traces, timeout=real_timeout
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/connector.py", line 548, in connect
    proto = await self._create_connection(req, traces, timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/connector.py", line 1056, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/connector.py", line 1411, in _create_direct_connection
    raise last_exc
  File "/usr/local/lib/python3.13/site-packages/aiohttp/connector.py", line 1380, in _create_direct_connection
    transp, proto = await self._wrap_create_connection(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<7 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/connector.py", line 1113, in _wrap_create_connection
    async with ceil_timeout(
               ~~~~~~~~~~~~^
        timeout.sock_connect, ceil_threshold=timeout.ceil_threshold
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/pypalazzetti/client.py", line 402, in _execute_command
    async with self._session.get(request_url) as response:
               ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client.py", line 1425, in __aenter__
    self._resp: _RetType = await self._coro
                           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/client.py", line 707, in _request
    raise ConnectionTimeoutError(
        f"Connection timeout to host {url}"
    ) from exc
aiohttp.client_exceptions.ConnectionTimeoutError: Connection timeout to host http://10.60.20.10/cgi-bin/sendmsg.lua?cmd=GET+STDT

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/palazzetti/config_flow.py", line 31, in async_step_user
    await client.connect()
  File "/usr/local/lib/python3.13/site-packages/pypalazzetti/client.py", line 48, in connect
    r = await self._execute_command(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<2 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pypalazzetti/client.py", line 409, in _execute_command
    raise CommunicationError("API communication error") from ex
pypalazzetti.exceptions.CommunicationError: API communication error
`````

Domochip, 2025-03-10:
That's interesting.
Could be a good idea that I give it a try because we are currently working on a reliability problem with @abcklaus .
To clarify things, the first post mention that this problem occurs when changing the temperature (setPoint) while the stove is off.
Does it also happens when the stove is on?

@dotvav  Even if this is something you can fix, I want to give it a try to understand if this is a memory shortage or a "simple" watchdog timeout. (I can improve things for watchdog timeout)

abcklaus, 2025-03-10:
Yes, the issue occurs only if the stove is turned off.

damien-louis, 2025-11-11:
Hi. Same behavior for me.

I can help you to debug if you need.