Domochip/WPalaControl #70: Pellet Consumed (PQT) is not updating

hmsjel, 2025-05-22:
Hi
I have bought a WPalaControl and had it set up in HomeAssistant
I can see that it cannot read the serial number and Pellet Consumed (PQT) is not updating.
Do you have any idea what could be wrong?

[image]

hmsjel, 2025-05-23:
Hi
Do you know how I reset it to zero and do you know what the risks might be with that?

Domochip, 2025-05-23:
To do it:
- I'll need the STDT of your stove, from this data, I'll be able to figure out the address in memory where the PQT should be.
- then you'll need to compile the firmware in developper mode, flash it on your module
- and throw an EXT+ADWR command to write the value 0 tothe correct address
- flash back the non developper firmware

About the risk, I definitely don't know. Worst case is stove malfunction.

Domochip, 2025-05-23:
Sorry, I don't need the STDT because there is only one address for PQT which is 0x2002

updated steps:
- you'll need to compile the firmware in developper mode, flash it on your module
- throw EXT+ADWR+2002+0+1
- flash back the non developper firmware

hmsjel, 2025-05-23:
Thanks for that, 
I'm considering whether to try it, I don't want to have to deal with a faulty pellet stove.

closed 2025-07-31 by Domochip