Domochip/WPalaControl #72: F4L does not appear in Home Assistant via MQTT

quenting57, 2025-07-22:
[image]
[image]

Hi,

My stove has a ducted output. When I physically change the fan speed of the ducted output on the stove, the F4L value updates correctly on the web interface.

However, F4L does not appear in Home Assistant via MQTT, unlike other controls such as temperature, power level, etc.

I’d like to integrate this F4L value into Home Assistant like the other MQTT entities.
Is it possible to expose it, or is there a specific configuration needed?

I’ve attached two screenshots:
	•	Home Assistant interface (F4L missing)
	•	Stove’s web interface (F4L visible and updating)

Thanks!

quenting57, 2025-07-28:
I have noticed that the F4L value is properly published in the MQTT topic PalazettiControl/FAND in JSON format, for example : 
`{"F1V":0,"F2V":0,"F1RPM":0,"F2L":2,"F2LF":0,"F3S":0.00,"F4S":0.00,"F3L":0,"F4L":5}`

quenting57, 2025-07-28:
Meanwhile, my Home Assistant solution :

> mqtt:
>   number:
>     - name: "Ducted outlet fan speed"
>       command_topic: "PalazettiControl/cmd"
>       command_template: "SET+FN4L+{{ value | int }}"
>       state_topic: "PalazettiControl/FAND"
>       value_template: "{{ value_json.F4L }}"
>       min: 0
>       max: 5
>       step: 1
>       unique_id: "ducted_outlet_fan_speed"

And synchronized with a Home Assistant input_number using automations.
**We need to find out why F4L wasn’t automatically discovered by Home Assistant**.

Domochip, 2025-07-31:
The only condition for FAN4 to appear is: FAN2TYPE > 2
This is true in your case, so something is weird here
if you connect to your MQTT broker using MQTT Explorer, do you find a retained message on a topic like : homeassistant/number/WPalaControl_LTXXXX_FAN4 OR homeassistant/switch/WPalaControl_LTXXXX_FAN4?

Domochip, 2025-12-23:
@quenting57 Could you compile the main branch and try it.
If you can't, let me know I can provide you a compiled version to flash.